Taeyoung96/GRIL-Calib #2: how to collect dataset better

ouguangjun, 2024-04-30:
What a fantastic project! and we conduct testing as soon as the code is released. 

Currently, I'm trying to use the Livox Mid-360 for testing. We haven't specifically collected datasets for calibration yet, but we're working with existing datasets. The convergence tends to be good in scenarios with more vigorous movements, but it's challenging in scenarios with slower speeds (less frequent turns). Even when convergence occurs, the results tend to be divergent, especially in terms of displacement. During data collection, do you have any suggestions for the vehicle's speed and the environment? We eagerly await your response.

Taeyoung96, 2024-04-30:
@ouguangjun   
Thanks for interesting our work!  
We've also found that performance is affected by the data collection process.  

We don't have a good standard for vehicle speed yet, but we recommend drawing an '8' so that the vehicle can move smoothly with enough movement.  

For slower speeds, try experimenting with lowering the thresholds for collections.  
For example, for velodyne16.yaml, you can lower the values of `data_accum_length: 300` and `z_accumulate : 0.99`.

ouguangjun, 2024-05-06:
Thanks for your reply and suggestion ：） 👍

closed 2024-05-06 by ouguangjun