Taeyoung96/GRIL-Calib #3: The estimated height fluctuates frequently

ouguangjun, 2024-05-06:
Hi @Taeyoung96 ,

Thanks for your great work! 
When I test the dataset we collected and find that the estimated height fluctuates frequently and is quite different from the value we measured manually. 

I try to visualize the "/patchworkpp/ground" and find that it not only contains the ground points but also the points on the ceiling, as shown the image bellow:
[image]
[image]

For the parameter for the patchworkpp in the config file , I just change the "sensor_height" and do not make any other changes.  Do I need to modify anything else? The height of the imu is approximately 0.22m and the height of the lidar is 0.075m.  Looking forward to your reply.  Thanks

ouguangjun, 2024-05-06:
I did a simple filtering on ground_points (removing ceiling point clouds) and found that it didn't seem to have any effect on the calibration results

Taeyoung96, 2024-05-07:
@ouguangjun   
Thanks for interesting our work!  

If the ground estimation is done well, the calibration results should look better.  
In your experiments, is the lidar odometry outputting well?  

Thanks,

ouguangjun, 2024-05-08:
@Taeyoung96 
Thanks for your reply. The rotation matrix (from ground to lidar/better) look more stable. I test GRIL-Calib and LI-init on the same sequence and find that gril-calib get better result on translation.

closed 2024-05-08 by ouguangjun