Taeyoung96/GRIL-Calib #5: x,y,z_accumulate and Impact on accuracy

momoistda, 2024-07-02:
Hello,
Thanks for posting this nice method!
I have two questions:

1. In the config fiel, we should give the x,y,z_accumulate:

x_accumulate: Parameter that determines how much the x-axis rotates (Assuming the x-axis is front)
y_accumulate: Parameter that determines how much the y-axis rotates (Assuming the y-axis is left)
z_accumulate: Parameter that determines how much the z-axis rotates (Assuming the z-axis is up)

**SO, If my y-axis is positiv to the right, should i fill in 0.999?**

**2. Are there any time limit and path requirements (like how lang should the path of this figure 8 shape be?)  for data collection? I used same robot with same speed, but output parameters are very different. See below, data1 and data2 were collected in same place:**

data1:
LiDAR-IMU calibration result:
Rotation LiDAR to IMU (degree)     = -3.664754  0.884811 -7.481670
Translation LiDAR to IMU (meter)   = -0.445991 -0.568093 -0.098269
Time Lag IMU to LiDAR (second)     = -2.458705
Bias of Gyroscope  (rad/s)         = -0.010000  0.010000  0.010000
Bias of Accelerometer (meters/s^2) = -0.008860  0.010584 -0.010463

Homogeneous Transformation Matrix from LiDAR frmae L to IMU frame I: 
 0.991370  0.128955  0.023600 -0.445991
-0.130184  0.989589  0.061363 -0.568093
-0.015441 -0.063906  0.997836 -0.098269
 0.000000  0.000000  0.000000  1.000000

data2:
LiDAR-IMU calibration result:
Rotation LiDAR to IMU (degree)     = -0.514068 -0.272551 10.003449
Translation LiDAR to IMU (meter)   =  0.279931 -1.056407 -0.092049
Time Lag IMU to LiDAR (second)     = 0.659137
Bias of Gyroscope  (rad/s)         = -0.010000  0.010000  0.002501
Bias of Accelerometer (meters/s^2) = -0.008049 -0.011665 -0.009957

Homogeneous Transformation Matrix from LiDAR frmae L to IMU frame I: 
 0.984788 -0.173646 -0.006242  0.279931
 0.173693  0.984767  0.008009 -1.056407
 0.004757 -0.008971  0.999948 -0.092049
 0.000000  0.000000  0.000000  1.000000

Taeyoung96, 2024-07-12:
@momoistda 
Thanks for interesting our work!  
1. If you're confused about the axes, we suggest experimenting with 0.999 for all `x_accumulate`, `y_accumulate`, and `z_accumulate` first, and then re-enter `x_accumulate`, `y_accumulate`, and `z_accumulate` using the percentages in the terminal when you're done collecting data.  
For example, if the output to the terminal at the end of a data collection is ""Rotation around Lidar X Axis: 70%", then the correct value for the data collection you have now is x_accumulate of 0.7.

2. The exact time to collect data was not covered in this research. Did you set your config file differently? The `gyro_factor`, `acc_factor`, and `ground_factor` can cause different results.

momoistda, 2024-07-25:
@Taeyoung96 
Thank you very much for your answer.
For 1: I use the GRIL-Calib for off-line calibration. For example, if I get the information from the terminal during the calibration like this:
          
[image]

That means, x_accumulate is 29% or 100%? Because it shows like "Rotation around Lidar X Axis complete!"

For 2. no, i just used the standard value from the velodyne16.yaml file.

Taeyoung96, 2024-09-08:
@momoistda   
I apologize for the late response.  

1. I think this is a bug in the visualization part, currently it's 29% based on the `data_accum_length` you set in the config file, but it says 100% in this scenario because it judges completion based on `x_accumulate` as a threshold. In other words, it looks like you've done the right thing for your experiment.  

2. It's a little confusing that the LiDAR sensor height is a negative value. I think you need to tune the parameter again so that the height of the LiDAR sensor is properly judged.  Also in my experience, it's a better way to experiment with slightly reducing the 'acc_factor'.

Taeyoung96, 2024-12-01:
It will be closed if no further activity occurs. 
Thanks 😄

closed 2024-12-01 by Taeyoung96