Taeyoung96/GRIL-Calib #10: Factors

jiheddachraoui, 2024-09-12:
hello how can i exactly choose gyro_factor, acc_factor, ground_factor.
Thank you

Taeyoung96, 2024-09-16:
@jiheddachraoui   
Thanks for interesting our works!  

We found the optimal parameter settings through experimentation.

jiheddachraoui, 2024-09-16:
@Taeyoung96 , thank you for your response. In my case, I am using an external, highly precise IMU rather than the one integrated into my Ouster lidar. Could this impact the results and the way I should configure the parameters?

Taeyoung96, 2024-09-19:
@jiheddachraoui     
Because the location of the IMU has changed, the results of the calibration will be different, but you can use the same pipeline for the calibration as long as you match the topic names.  

Do you want to utilize the LIO as a result of the calibration parameter?  
If so, I think you could probably get away with using ouster's internal IMU.

Taeyoung96, 2024-12-01:
It will be closed if no further activity occurs.
Thanks 😄

closed 2024-12-01 by Taeyoung96