Taeyoung96/GRIL-Calib #12: how to process S3E db3 data

huanhexiao, 2024-10-18:
As the title, I want to inquery that how do you process .db3 data?

huanhexiao, 2024-10-18:
@Taeyoung96

Taeyoung96, 2024-10-21:
@huanhexiao   
Thanks for interesting our work!  

I've been experimenting with S3E on the ros1 version.  
I am guessing that the dataset in S3E has been updated.

If this is the case, you can use [Rosbags](https://pypi.org/project/rosbags/) to convert the dataset to ros1 bag format or try our [humble](https://github.com/Taeyoung96/GRIL-Calib/tree/humble?tab=readme-ov-file) branch! (humble version is not fully tested yet)

huanhexiao, 2024-10-25:
@Taeyoung96 
Thank you for your reply.
I've tried "rosbag info S3E_xxx.db3", "rosbag play S3E_xxx.db3", they both failed. 
As I'm not very familiar with ROS, could u offer me something on this problem ?

Taeyoung96, 2024-10-26:
@huanhexiao   
The .db3 format is a data format that can be utilized by ROS2.
I recommend that you refer to the [this documentation](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Recording-And-Playing-Back-Data/Recording-And-Playing-Back-Data.html).

huanhexiao, 2024-10-29:
@Taeyoung96 
It works with both ROS2 and your [humble](https://github.com/Taeyoung96/GRIL-Calib/tree/humble?tab=readme-ov-file) branch!
Thank you Taeyoung!

closed 2024-10-29 by Taeyoung96