Taeyoung96/GRIL-Calib #14: Other types of Spinning LiDAR

mainDevgithub, 2024-11-30:
Is it possible to use other types of Spinning LiDAR (e.g. lslidar-C16 (镭神-C16))? Or what kind of adaptation is needed?

Taeyoung96, 2024-12-01:
@mainDevgithub 
Thanks for interesting our work!   

Unfortunately, we don't have any tests for robosense lidar at this time. 

We think that the robosense lidar type requires an additional implementation.  
This can be implemented by referring to the link below.  

[LI-Init/preprocess.h for Robosense](https://github.com/hku-mars/LiDAR_IMU_Init/blob/c34506f9e59fb67e126ce96d802abcefe8eb44d5/src/preprocess.h#L97)  
[LI-Init/preprocess.cpp for Robosense](https://github.com/hku-mars/LiDAR_IMU_Init/blob/c34506f9e59fb67e126ce96d802abcefe8eb44d5/src/preprocess.cpp#L234)

Taeyoung96, 2024-12-15:
@mainDevgithub   
As it mentioned at #17, you could use https://github.com/HViktorTsoi/rs_to_velodyne repository.

mainDevgithub, 2024-12-15:
Perhaps I should have replied earlier. My lidar may not be common in open source projects, it is ls-lidar(leishen) instead of rs-lidar(robosense).
Still thank you very much for your reply.

lostDeers, 2025-04-13:
Hello @mainDevgithub 
I meet some adaption problems with lslidar c16. How did you solve these problems? Appreciate for your support.

mainDevgithub, 2025-04-13:
> Hello @mainDevgithub 
> I meet some adaption problems with lslidar c16. How did you solve these problems? Appreciate for your support.

Could you describe the issue in more detail? I'll see if I can assist you.

We can talk in Chinese if that's easier for you.

lostDeers, 2025-04-14:
感谢您的回复。我正在使用镭神-C16进行标定测试，在标定过程中我发现，`Data accumulation`阶段总是会瞬间结束，并且在之后给出了错误的结果（或者说是无结果），我的log如下（其中用户名已用`{user}`代替），对config的修改仅有修改激光线数为`16`
```
{user}@ubuntu:~/calib-tools/gril$ ros2 launch gril_calib mapping_velodyne.launch.py 
[INFO] [launch]: All log files can be found below /home/{user}/.ros/log/2025-04-13-12-48-47-660341-ubuntu-63786
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gril_calib_exec-1]: process started with pid [63793]
[INFO] [rviz2-2]: process started with pid [63795]
[gril_calib_exec-1] Multi thread started 
[gril_calib_exec-1] lidar_type: 2
[gril_calib_exec-1] LiDAR-only odometry starts.
[gril_calib_exec-1] Operating patchwork++...
[gril_calib_exec-1] [INFO 1744519727.913783744] [patchworkpp]: Inititalizing PatchWork++... (PatchWorkpp() at /home/{user}/calib-tools/gril/GRIL-Calib/include/GroundSegmentation/patchworkpp.hpp:85)
[gril_calib_exec-1] Sensor Height: 0.6
[gril_calib_exec-1] Num of Iteration: 3
[gril_calib_exec-1] Num of LPR: 20
[gril_calib_exec-1] Num of min. points: 10
[gril_calib_exec-1] Seeds Threshold: 0.3
[gril_calib_exec-1] Distance Threshold: 0.125
[gril_calib_exec-1] Max. range:: 4
[gril_calib_exec-1] Min. range:: 1
[gril_calib_exec-1] Normal vector threshold: 0.707
[gril_calib_exec-1] adaptive_seed_selection_margin: -1.2
[gril_calib_exec-1] Num. zones: 4
[gril_calib_exec-1] Num. sectors: 16, 32, 54, 32
[gril_calib_exec-1] Num. rings: 3, 3, 3, 3
[gril_calib_exec-1] elevation_thr_: 0.1000, 0.2000, 0.4000, 0.6000 
[gril_calib_exec-1] flatness_thr_: 0.0000, 0.0000, 0.0000, 0.0000 
[gril_calib_exec-1] INITIALIZATION COMPLETE
[gril_calib_exec-1] ~~~~/home/{user}/calib-tools/gril/GRIL-Calib/ doesn't exist
[gril_calib_exec-1] [INFO 1744519727.932033167] [laser_mapping]: Node init finished. (LaserMappingNode() at /home/{user}/calib-tools/gril/GRIL-Calib/src/laserMapping.cpp:1049)
[rviz2-2] [INFO 1744519728.408403931] [rviz2]: Stereo is NOT SUPPORTED (operator()() at ./src/main.cpp:62)
[rviz2-2] [INFO 1744519728.408630786] [rviz2]: OpenGl version: 4.6 (GLSL 4.6) (operator()() at ./src/main.cpp:62)
[rviz2-2] [INFO 1744519728.436635083] [rviz2]: Stereo is NOT SUPPORTED (operator()() at ./src/main.cpp:62)
[gril_calib_exec-1] [Initialization] Movement detected, data accumulation starts.
[gril_calib_exec-1] 
[gril_calib_exec-1] 
[gril_calib_exec-1] 
[gril_calib_exec-1] 
[gril_calib_exec-1] Delete time: 0.000002, delete size: 0
[gril_calib_exec-1] 
[Data accumulation] Rotation around Lidar X Axis complete! 
[gril_calib_exec-1] [Data accumulation] Rotation around Lidar Y Axis complete! 
[gril_calib_exec-1] [Data accumulation] Rotation around Lidar Z Axis complete! 
[gril_calib_exec-1] [Rotation matrix Ground to LiDAR (euler)]  0 -0  0 deg
[gril_calib_exec-1] [Rotation matrix Ground to IMU (euler)]  0 -0  0 deg
[gril_calib_exec-1] [Estimated LiDAR sensor height] : 0 m
[ERROR] [gril_calib_exec-1]: process has died [pid 63793, exit code -11, cmd '/home/{user}/calib-tools/gril/install/gril_calib/lib/gril_calib/gril_calib_exec --ros-args --params-file /home/{user}/calib-tools/gril/install/gril_calib/share/gril_calib/config/c16.yaml --params-file /tmp/launch_params_4dj207bc'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rviz2-2] [INFO 1744519813.532403214] [rclcpp]: signal_handler(signum=2) (signal_handler() at ./src/rclcpp/signal_handler.cpp:71)
[INFO] [rviz2-2]: process has finished cleanly [pid 63795]

```

lostDeers, 2025-04-14:
请问您有试过其他的标定方法吗？我尝试过[OA-LICalib](https://github.com/APRIL-ZJU/OA-LICalib)，但是也是出现了无法正常初始化的结果。
我猜测可能是驱动的不同带来的影响，请问您当时使用的是哪一款驱动？

mainDevgithub, 2025-04-14:
我当时是在 ros melodic 上运行的，激光雷达的驱动是机器人厂家弄好的。后来我在做其它事情，到目前为止我还没用其它算法进行标定

lostDeers, 2025-04-14:
请问可以分享您的驱动吗？我想对比一下代码方面有何不同

lostDeers, 2025-04-14:
请问可以分享一下您的驱动吗？我想对比一下代码有何不同。

mainDevgithub, 2025-04-14:
你可以留个邮箱之类的，我抽空发给你

lostDeers, 2025-04-15:
我的邮箱是公开的，麻烦您了