Taeyoung96/GRIL-Calib #15: Question Regarding Calibration Error RMSE Results in the Paper

Guoc0529, 2024-12-01:
First of all, I would like to thank the author(s) for their open-source contribution. I have a question regarding the calibration experiments and comparison results presented in the paper. Specifically, I noticed the RMSE calibration error results for several algorithms. I would like to know how the "ground truth" calibration results were obtained for the comparison. Could you please provide more details on how the real calibration results were derived?

Thank you in advance for your time and support!

Taeyoung96, 2024-12-01:
@Guoc0529   
Thanks for interesting our work!  

For the public datasets, we used the calibration values provided by the dataset for the ground truth values.  
For the custom datasets in the Supplementatry Material, we used ground truth values based on CAD modeling.  

For each ground truth value, we calculated the RMSE by utilizing the respective values of the measurements from each algorithm.  

We hope this answer is helpful.

Guoc0529, 2024-12-02:
> @Guoc0529 Thanks for interesting our work!感谢您感兴趣的工作！
> 
> For the public datasets, we used the calibration values provided by the dataset for the ground truth values.对于公共数据集，我们使用数据集提供的校准值作为真值。 For the custom datasets in the Supplementatry Material, we used ground truth values based on CAD modeling.对于补充材料中的自定义数据集，我们使用了基于 CAD 建模的真值。
> 
> For each ground truth value, we calculated the RMSE by utilizing the respective values of the measurements from each algorithm.对于每个真值，我们利用每种算法的相应测量值来计算 RMSE。
> 
> We hope this answer is helpful.我们希望这个答案对您有所帮助。

Alright, I understand now. Thank you for your answer!

closed 2024-12-02 by Taeyoung96