Taeyoung96/GRIL-Calib #17: Robosen lidar calib result ！！！

MrDragon2, 2024-12-03:
Hi author,
 I calibrated with Robosen radar and IMU, but the result is incorrect：
      LiDAR-IMU calibration result:
      Rotation LiDAR to IMU (degree)     =   0.001848  -0.314683 109.471282
      Translation LiDAR to IMU (meter)   =  0.131908 -0.216292 -0.840857
      Time Lag IMU to LiDAR (second)     = -0.161370
      Bias of Gyroscope  (rad/s)         = -0.000975 -0.001078  0.000365
      Bias of Accelerometer (meters/s^2) = -0.006084 -0.012796  0.006816
      
      Homogeneous Transformation Matrix from LiDAR frmae L to IMU frame I: 
      -0.333197 -0.942856  0.001860  0.131908
       0.942841 -0.333202 -0.005167 -0.216292
       0.005492  0.000032  0.999985 -0.840857
       0.000000  0.000000  0.000000  1.000000
What is the problem with the obvious incorrect result of this rotation and translation matrix?

MrDragon2, 2024-12-04:
[Data accumulation] Rotation around Lidar X Axis complete! 
[Data accumulation] Rotation around Lidar Y Axis complete! 
[Data accumulation] Rotation around Lidar Z Axis:  85% [|||||||||||||||||||||||||     ]
[Rotation matrix Ground to LiDAR (euler)]     0.1033      -1.08 -0.0009733 deg
[Rotation matrix Ground to IMU (euler)] 0.09112 -0.4713   91.48 deg
[Estimated LiDAR sensor height] : 0.3132 m
[laserMapping-3] process has died [pid 1713835, exit code -11, cmd /home/lgg/catkin_ws/devel/lib/gril_calib/gril_calib __name:=laserMapping __log:=/home/lgg/.ros/log/b704e392-b20c-11ef-9c2d-1f1a0c2c2551/laserMapping-3.log].
log file: /home/lgg/.ros/log/b704e392-b20c-11ef-9c2d-1f1a0c2c2551/laserMapping-3*.log

The program crashed！！！？？？？

MrDragon2, 2024-12-05:
> [Data accumulation] Rotation around Lidar X Axis complete! [Data accumulation] Rotation around Lidar Y Axis complete! [Data accumulation] Rotation around Lidar Z Axis: 85% [||||||||||||||||||||||||| [...]

I am using lidar is Helios 32

Taeyoung96, 2024-12-06:
@MrDragon2 

Thanks for interesting our work!  

1. The calibration results will depend on how you acquired the data: you should utilize figure-of-eight motion to acquire the data, and make sure that the ground is reliably estimated. You should also check parameters in the config file, such as the height of the IMU, LiDAR height, etc.  

2. Unfortunately, I haven't experimented with Helios 32 yet, so I'm curious to know what modifications you've made and what results you've seen. Does it always crash? Or does it only happen occasionally?

MrDragon2, 2024-12-09:
> @MrDragon2
>
> Thanks for interesting our work!
>
> 1. The calibration results will depend on how you acquired the data: you should utilize figure-of-eight motion to acquire the data, and make sure that [...]

The sensor parameters you mentioned have already been set, and I will reduce the parameter 'z'accumulate' and it will not crash；z_accumulate : 0.3 #0.99

But the result still has a large margin of error：
Homogeneous Transformation Matrix from LiDAR frmae L to IMU frame I: 
-0.234803 -0.972043  0.000229 -0.046851
 0.972032 -0.234801 -0.004810 -0.374820
 0.004729 -0.000907  0.999988 -0.841409
 0.000000  0.000000  0.000000  1.000000

MrDragon2, 2024-12-09:
> I am using lidar is Helios 32

I use the function package rs_to_velodyne to convert Robosen's data into velodyne.

Taeyoung96, 2024-12-15:
@MrDragon2 

If you have a reliable initial value for the translation vector, we recommend replacing `set_boundary : true``bound_th : 0.1` in the yaml file and experimenting.

MrDragon2, 2024-12-17:
> @MrDragon2 z_accumulate is simply a threshold to accumulate data. If you have a reliable initial value for the translation vector, we recommend replacing `` set_boundary : true``bound_th : 0.1 `` in the yaml file and experimenting.

Homogeneous Transformation Matrix from LiDAR frmae L to IMU frame I: 
-0.276476 -0.961020 -0.000955 -0.068680
 0.960989 -0.276459 -0.008391 -0.125725
 0.007800 -0.003237  0.999964  0.234888
 0.000000  0.000000  0.000000  1.000000  
The result is incorrect.......