Taeyoung96/GRIL-Calib #19: Some question about the Jacobian

q87956256, 2024-12-06:
Thank you for your wonderful work!
I have two questions about the code:

1.In formula 8, the results from the code corresponding to the Jacobian derivation of the error with respect to the error state variables are different from those I derived.
In your code, the Jacobian of the error with respect to the error state variables is:
[image]
However, my derivation process and results are shown in the figure below:
[image]
That is different with your code.
Could you kindly share your derivation process?

2.In formula 9, I have the same question about the Jacobian:
In your code, the Jacobian of the error with respect to the error state variables is:
[image]
However, my derivation process and results are shown in the figure below:
[image]
That is different with your code.
Could you kindly share your derivation process?

Taeyoung96, 2024-12-15:
@q87956256   
Thanks for interesting our work!  

I went back through the derivation process and it seems that I made a terrible mistake in my approch: I made a mistake when I did the perturbation on the rotation matrix.  

If I rearrange it, it looks like this,

---

### 1. State Vector
The state vector \(\mathbf{x}_k\) is defined as:
\[
\mathbf{x}_k = [\mathbf{R}_{L_k}, \mathbf{p}_{L_k}, \boldsymbol{\omega}_{L_k}, \mathbf{v}_{L_k}]
\]
where:
- \(\mathbf{R}_{L_k} \in SO(3)\): Rotation matrix
- \(\mathbf{p}_{L_k} \in \mathbb{R}^3\): Position
- \(\boldsymbol{\omega}_{L_k} \in \mathbb{R}^3\): Angular velocity
- \(\mathbf{v}_{L_k} \in \mathbb{R}^3\): Linear velocity

### 2. Residual \(\mathbf{z}_{\pi}\)
The residual is defined as:
\[
\mathbf{z}_{\pi} =
\begin{pmatrix}
\left[ (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \cdot \mathbf{n}_{L_k} \right]_{1,2} \\ \\
\mathbf{e}_3^{\mathbf{T}} \cdot (\mathbf{R}^G_L)^T \mathbf{p}_{L_k}
\end{pmatrix}
\]
where:
- \(\mathbf{R}^G_L\): Fixed reference rotation matrix
- \(\mathbf{n}_{L_k} \in \mathbb{R}^3\): Specific vector
- \(\mathbf{e}_3 = [0, 0, 1]^{\mathbf{T}}\): Unit vector along the \(z\)-axis

### 3. Goal
The goal is to compute the Jacobian matrix:
\[
\mathbf{H}_{\pi} = \left.\frac{\partial \mathbf{h}_{\pi}(\hat{\mathbf{x}} \boxplus \tilde{\mathbf{x}}) \boxminus \mathbf{z}_{\pi}}{\partial \tilde{\mathbf{x}}}\right|_{\tilde{\mathbf{x}}=\mathbf{0}}
\]
where:
- \(\hat{\mathbf{x}} \boxplus \tilde{\mathbf{x}}\): Perturbed state
- \(\tilde{\mathbf{x}}\): Perturbation vector

---

## Derivation of Jacobian Matrix

### 1. Perturbing the State Rotation
The rotation matrix \(\mathbf{R}_{L_k} \in SO(3)\) is perturbed using a small rotation vector \(\tilde{\boldsymbol{\theta}}\), which lies in the tangent space of \(SO(3)\). The perturbation is represented using the exponential map:
\[
\mathbf{R}_{L_k} \rightarrow \mathbf{R}_{L_k} \exp([\tilde{\boldsymbol{\theta}}]_{\times}).
\]
Here:
- \([\tilde{\boldsymbol{\theta}}]_{\times} \in \mathbb{R}^{3 \times 3}\): Skew-symmetric matrix corresponding to \(\tilde{\boldsymbol{\theta}}\).

For small perturbations, the exponential map can be approximated as:
\[
\exp([\tilde{\boldsymbol{\theta}}]_{\times}) \approx \mathbf{I} + [\tilde{\boldsymbol{\theta}}]_{\times}.
\]

Thus, the perturbed rotation matrix becomes:
\[
\mathbf{R}_{L_k} \exp([\tilde{\boldsymbol{\theta}}]_{\times}) \approx \mathbf{R}_{L_k} (\mathbf{I} + [\tilde{\boldsymbol{\theta}}]_{\times}).
\]

---

### 2. Impact of Perturbation on Residual \(\mathbf{z}_{\pi,1}\)
The first component of the residual is:
\[
\mathbf{z}_{\pi,1} = \left[ (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \mathbf{n}_{L_k} \right]_{1,2}.
\]

1. **Rotation Perturbation**: Under perturbation, the product \((\mathbf{R}^G_L)^T \mathbf{R}_{L_k}\) becomes:
   \[
   (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \rightarrow (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} (\mathbf{I} + [\tilde{\boldsymbol{\theta}}]_{\times}).
   \]

2. **Vector Perturbation**: The term involving \(\mathbf{n}_{L_k}\) transforms as:
   \[
   (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} (\mathbf{I} + [\tilde{\boldsymbol{\theta}}]_{\times}) \mathbf{n}_{L_k}.
   \]

Expanding this to first order:
\[
(\mathbf{R}^G_L)^T \mathbf{R}_{L_k} (\mathbf{I} + [\tilde{\boldsymbol{\theta}}]_{\times}) \mathbf{n}_{L_k} \approx (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \mathbf{n}_{L_k} + (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\tilde{\boldsymbol{\theta}}]_{\times} \mathbf{n}_{L_k}.
\]

The perturbation introduced in \(\mathbf{z}_{\pi,1}\) is therefore:
\[
\Delta \mathbf{z}_{\pi,1} \approx \left[ (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\tilde{\boldsymbol{\theta}}]_{\times} \mathbf{n}_{L_k} \right]_{1,2}.
\]

---

### 3. Interpreting the Skew-Symmetric Term
The skew-symmetric matrix satisfies the property:
\[
[\tilde{\boldsymbol{\theta}}]_{\times} \mathbf{v} = - \mathbf{v} \times \tilde{\boldsymbol{\theta}} \quad \text{for any vector } \mathbf{v}.
\]

Substituting this into the previous expression:
\[
(\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\tilde{\boldsymbol{\theta}}]_{\times} \mathbf{n}_{L_k} = - (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} (\mathbf{n}_{L_k} \times \tilde{\boldsymbol{\theta}}).
\]

Using the property of the cross product:
\[
\mathbf{n}_{L_k} \times \tilde{\boldsymbol{\theta}} = [\mathbf{n}_{L_k}]_{\times} \tilde{\boldsymbol{\theta}}.
\]

Thus:
\[
(\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\tilde{\boldsymbol{\theta}}]_{\times} \mathbf{n}_{L_k} = - (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\mathbf{n}_{L_k}]_{\times} \tilde{\boldsymbol{\theta}}.
\]

---

### 4. Jacobian with Respect to \(\tilde{\boldsymbol{\theta}}\)
The perturbation \(\Delta \mathbf{z}_{\pi,1}\) becomes:
\[
\Delta \mathbf{z}_{\pi,1} \approx \left[ - (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} [\mathbf{n}_{L_k}]_{\times} \tilde{\boldsymbol{\theta}} \right]_{1,2}.
\]

The Jacobian with respect to \(\tilde{\boldsymbol{\theta}}\) is:
\[
\frac{\partial \mathbf{z}_{\pi,1}}{\partial \tilde{\boldsymbol{\theta}}} = \left[ (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \mathbf{n}_{L_k} \right]_{\times, 1,2}.
\]

---

## Final Jacobian Matrix
Combining all terms, the Jacobian matrix \(\mathbf{H}_{\pi}\) is:
\[
\mathbf{H}_{\pi} =
\begin{bmatrix}
\left[ (\mathbf{R}^G_L)^T \mathbf{R}_{L_k} \mathbf{n}_{L_k} \right]_{\times, 1,2} & \mathbf{0} & \mathbf{0} & \mathbf{0} \\
\mathbf{0} & \mathbf{e}_3^{\mathbf{T}} \cdot (\mathbf{R}^G_L)^T & \mathbf{0} & \mathbf{0}
\end{bmatrix}.
\]

q87956256, 2024-12-16:
Thanks for your reply!
Your Jacobian derivation method and results align with mine, which is reassuring.
The last thing I'd like to confirm is:
In your derivation:
[image]
Is  [image] in your derivation equivalent to [image] in Formula (10) from your paper? 
[image]

Looking forward to your clarification!
Best regards,