Taeyoung96/GRIL-Calib #23: Ouster OS-1 128

BasuPrithviraj, 2025-02-12:
Hi, first of all great work with the paper! I am working with an ouster 128, and I need some help with the config file. I tried to do it on my own but I am getting the error:

`auto-starting new master
process[master]: started with pid [18249]
ROS_MASTER_URI=http://gaiaxamsubuntu-Amd-Wrx80:11311

setting /run_id to 735661f4-e932-11ef-9d7b-50849245c9ff
process[rosout-1]: started with pid [18260]
started core service [/rosout]
process[laserMapping-2]: started with pid [18267]
Multi thread started 
lidar_type: 3
LiDAR-only odometry starts.
Operating patchwork++...
[ INFO] [1739358837.670649925]: Inititalizing PatchWork++...
[ INFO] [1739358837.675443245]: Sensor Height: 2.000000
[ INFO] [1739358837.675459435]: Num of Iteration: 3
[ INFO] [1739358837.675466128]: Num of LPR: 20
[ INFO] [1739358837.675474503]: Num of min. points: 5
[ INFO] [1739358837.675481897]: Seeds Threshold: 0.300000
[ INFO] [1739358837.675489151]: Distance Threshold: 0.125000
[ INFO] [1739358837.675495913]: Max. range:: 5.000000
[ INFO] [1739358837.675508346]: Min. range:: 1.000000
[ INFO] [1739358837.675515640]: Normal vector threshold: 0.707000
[ INFO] [1739358837.675522994]: adaptive_seed_selection_margin: -1.200000
[ INFO] [1739358837.676222946]: Num. zones: 4
Num. sectors: 16, 32, 54, 32
Num. rings: 2, 4, 4, 4
elevation_thr_: 0.1000, 0.2000, 0.4000, 0.6000 
flatness_thr_: 0.0000, 0.0000, 0.0000, 0.0000 
INITIALIZATION COMPLETE
~~~~/root/catkin_ws/src/ doesn't exist
process[rviz-3]: started with pid [18268]
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Failed to find match for field 't'.
Failed to find match for field 'reflectivity'.
Failed to find match for field 'ring'.
Failed to find match for field 'ambient'.
Failed to find match for field 'range'.
[laserMapping-2] process has died [pid 18267, exit code -11, cmd /root/catkin_ws/devel/lib/gril_calib/gril_calib __name:=laserMapping __log:=/root/.ros/log/735661f4-e932-11ef-9d7b-50849245c9ff/laserMapping-2.log].
log file: /root/.ros/log/735661f4-e932-11ef-9d7b-50849245c9ff/laserMapping-2*.log
`

I recorded the bag at 1024X10 with xyzir mode

zqyhia, 2025-05-21:
Hello, have you figured out why accumulation wasn't starting? I met same problem while using ros2bag to test GRIL-Calib.

BasuPrithviraj, 2025-05-21:
@zqyhia , yes I did. I set the x and y accumulation to 0.01 and z to 0.99. 

Then during the data collection (recording the ros2 bag), after starting the recording, I drove 1 meter forward with high acceleration. Then stopped the car for a couple of seconds, this usually was enough to force the algorithm to make it start it's detection. 

Then just continue your recording just as he had mentioned...

closed 2025-05-21 by BasuPrithviraj

zqyhia, 2025-05-21:
> [@zqyhia](https://github.com/zqyhia) , yes I did. I set the x and y accumulation to 0.01 and z to 0.99.
> 
> Then during the data collection (recording the ros2 bag), after starting the recording, I drove 1 meter forward with high acceleration. Then stopped the car for a couple of seconds, this usually was enough to force the algorithm to make it start it's detection.
> 
> Then you may just continue your recording just as he had mentioned in the 8 shape...

Thank you very much!! Although I can't try your solution immediately, I think it should be able to solve this problem very well. I will make an attempt later. Thanks again!!

BasuPrithviraj, 2025-05-21:
Welcome 😊